nvidia-holoscan/holoscan-sdk #4: Failed to run sample | Failed to create the Vulkan instance.

AbdelkrimBLab, 2023-01-20:
Hi, I am trying to run :  python3 ./apps/endoscopy_tool_tracking/python/endoscopy_tool_tracking.py --source=replayer using docker in x86 Platform, I got this error:
Vulkan Version:
 - available:  1.2.131
 - requesting: 1.2.0
VK_ERROR_EXTENSION_NOT_PRESENT: VK_KHR_external_memory_capabilities - 0
2023-01-20 10:56:08.963 ERROR /workspace/holoscan-sdk/gxf_extensions/holoviz/holoviz.cpp@513: Failed to create the Vulkan instance.
2023-01-20 10:56:08.963 WARN  gxf/std/greedy_scheduler.cpp@235: Error while executing entity 42 named 'holoviz': GXF_FAILURE
2023-01-20 10:56:08.966 ERROR gxf/std/entity_executor.cpp@540: Entity [holoviz] must be in Lifecycle::kStarted or Lifecycle::kIdle stage before stopping. Current state is StartPending
2023-01-20 10:56:08.966 INFO  gxf/std/greedy_scheduler.cpp@367: Scheduler finished.
2023-01-20 10:56:08.966 ERROR gxf/std/program.cpp@497: wait failed. Deactivating...
2023-01-20 10:56:08.967 DEBUG gxf/cuda/cuda_stream.cpp@62: CudaStream destroyed
2023-01-20 10:56:08.967 ERROR gxf/core/runtime.cpp@1251: Graph wait failed with error: GXF_FAILURE
2023-01-20 10:56:08.967 PANIC /workspace/holoscan-sdk/src/core/executors/gxf/gxf_executor.cpp@278: GXF operation failed: GXF_FAILURE

any ideas?
Thanks

AndreasHeumann, 2023-01-20:
Hi,
it seems there is no Vulkan device available for Holoviz to use.
How did you run the docker container?
For Vulkan to work the `/usr/share/vulkan/icd.d/nvidia_icd.json` or `/etc//vulkan/icd.d/nvidia_icd.json` file needs to mapped to the container. If you use `./run launch` this will be done by the script.

ABE-2011, 2023-01-20:
Thanks! it works

closed 2023-01-20 by ABE-2011