nvidia-holoscan/holoscan-sdk #7: Building on local x86_64 System

Brenaren, 2023-06-06:
I am trying to build the SDK on a Ubuntu 22.04.2 LTS Linux system. I am using the './run build' command from the public directory and get the error 
"--2023-06-06 16:46:34--  https://api.ngc.nvidia.com/v2/resources/nvstaging/holoscan/gxf_x86_64_holoscan_sdk/versions/23.04-57e63aaf/zip
Resolving api.ngc.nvidia.com (api.ngc.nvidia.com)... 34.223.181.98, 44.226.83.234
Connecting to api.ngc.nvidia.com (api.ngc.nvidia.com)|34.223.181.98|:443... connected.
HTTP request sent, awaiting response... 404 
2023-06-06 16:46:34 ERROR 404: (no description)."

When I use './run install_gxf' I get the same thing, which I assume means that is the place the issue is coming from. Is there a work around for this?

jjomier, 2023-06-07:
Which version of Holoscan SDK are you trying to build? Can you try with the latest main branch or v0.5.1 tag and see if you get the same issue. Also, you might want to try to get the [Debian packages](https://catalog.ngc.nvidia.com/orgs/nvidia/teams/clara-holoscan/resources/holoscan_dev_deb/files?version=v0.5.1-amd64) if you want the pre-built SDK.

Note also that we are not supporting 22.04 and Holsocan SDK has not been tested on this platform.

Brenaren, 2023-06-07:
Installing Ubuntu 20.04 on a VM and building it there fixed the issue. Thank you!

closed 2023-06-07 by Brenaren