nvidia-holoscan/holoscan-sdk #9: Building from source: Version '2:1.6.9-2ubuntu1.2' for 'libx11-dev' was not found

jokla, 2023-06-18:
Hi! 

I encountered this error while building the SKD with the `run` script.

```
Version '2:1.6.9-2ubuntu1.2' for 'libx11-dev' was not found
```

It looks like the  'libx11-dev' version `2:1.6.9-2ubuntu1.2` was replaced by `2:1.6.9-2ubuntu1.5` recently : https://launchpad.net/ubuntu/focal/amd64/libx11-dev

To fix the issue, I updated the version of `libx11-dev` to `2:1.6.9-2ubuntu1.5`  in the Dockerfile:
https://github.com/nvidia-holoscan/holoscan-sdk/blob/db666c3a516beeb3791b7ebd12f028f7c6547059/Dockerfile#L140

I have opened an issue since direct merging is not enabled in this repository.

Thanks!

jjomier, 2023-06-20:
Thank you for the report and fix Giovanni.  We are looking into this.

agirault, 2023-06-21:
Thank you again @jokla. Addressed in https://github.com/nvidia-holoscan/holoscan-sdk/commit/5691e14e2d9379336c51ddf31aef3c85a22d5947, and marked you as Co-Author.

closed 2023-06-21 by agirault