nvidia-holoscan/holoscan-sdk #10: Compilation error on GCC-13.

luigifcruz, 2023-06-21:
I'm encountering a CMake error when I try to compile this repository from the source using GCC-13. Compilation works when using GCC-12 `CC=gcc-12 CXX=g++-12 cmake ..`. I attached part of the error logs below. I know CUDA 12.1 does not support GCC-13 officially and it's a pretty new release but I'm filling this issue just as a reference for people with the same problem.

<details>

<summary>Compilation Log</summary>

```
  /usr/include/bits/mathcalls.h(71): error: identifier "_Float32" is
  undefined

     extern _Float32 coshf32 (_Float32 __x) noexcept (true); extern _Float32 __coshf32 (_Float32 __x) noexcept (true);
                              ^

  /usr/include/bits/mathcalls.h(71): error: identifier "_Float32" is
  undefined

     extern _Float32 coshf32 (_Float32 __x) noexcept (true); extern _Float32 __coshf32 (_Float32 __x) noexcept (true);
                                                                    ^

  /usr/include/bits/mathcalls.h(71): error: identifier "_Float32" is
  undefined

     extern _Float32 coshf32 (_Float32 __x) noexcept (true); extern _Float32 __coshf32 (_Float32 __x) noexcept (true);
                                                                                        ^

  /usr/include/bits/mathcalls.h(73): error: identifier "_Float32" is
  undefined

     extern _Float32 sinhf32 (_Float32 __x) noexcept (true); extern _Float32 __sinhf32 (_Float32 __x) noexcept (true);
            ^

  /usr/include/bits/mathcalls.h(73): error: identifier "_Float32" is
  undefined

     extern _Float32 sinhf32 (_Float32 __x) noexcept (true); extern _Float32 __sinhf32 (_Float32 __x) noexcept (true);
                              ^

  /usr/include/bits/mathcalls.h(73): error: identifier "_Float32" is
  undefined

     extern _Float32 sinhf32 (_Float32 __x) noexcept (true); extern _Float32 __sinhf32 (_Float32 __x) noexcept (true);
                                                                    ^

  /usr/include/bits/mathcalls.h(73): error: identifier "_Float32" is
  undefined

     extern _Float32 sinhf32 (_Float32 __x) noexcept (true); extern _Float32 __sinhf32 (_Float32 __x) noexcept (true);
                                                                                        ^

  /usr/include/bits/mathcalls.h(75): error: identifier "_Float32" is
  undefined

     extern _Float32 tanhf32 (_Float32 __x) noexcept (true); extern _Float32 __tanhf32 (_Float32 __x) noexcept (true);
            ^

  /usr/include/bits/mathcalls.h(75): error: identifier "_Float32" is
  undefined

     extern _Float32 tanhf32 (_Float32 __x) noexcept (true); extern _Float32 __tanhf32 (_Float32 __x) noexcept (true);
                              ^
```

</details>

jjomier, 2023-06-22:
Thanks Luigi for the report.

ichergui, 2024-08-02:
Hi @luigifcruz 
kindly check the online documentation for CUDA Toolkit `12.2` system requirements.
https://docs.nvidia.com/cuda/archive/12.2.2/cuda-installation-guide-linux/index.html#system-requirements
The recommendation is to use `GCC` version `11.4` 

Based on my humble experience with CUDA and as you mentioned. we can use `GCC` version `12.x` to satisfy the `NVCC` (Nvidia CUDA Compiler) requirements.
Kindly check the following link:
https://docs.nvidia.com/cuda/archive/12.2.2/cuda-installation-guide-linux/index.html#host-compiler-support-policy

Please let me know if you need any additional information.

tbirdso, 2024-08-02:
Thank you @ichergui . Moving to close as not planned to fix in Holoscan SDK.

closed 2024-08-02 by tbirdso