nvidia-holoscan/holoscan-sdk #15: Random segfault possibly related to shm conflicts?

genevanmeter, 2023-11-17:
AGX Orin 64GB igpu 
Jetpack 5.1.1
Running in docker (not a distributed application)
Python

This crash randomly occurs and is difficult to reliably reproduce.  It may be related to the use of eCAL (https://github.com/eclipse-ecal/ecal) as sometimes it can be trigger when eCAL monitor is subscribed to a topic which is being sent over the host ipc and shared memory. 

`[app:96   :0:131] Caught signal 11 (Segmentation fault: address not mapped to object at address 0xb8)
==== backtrace (tid:    131) ====
 0  /opt/nvidia/holoscan/python/lib/holoscan/graphs/../../../../lib/libucs.so.0(ucs_handle_error+0x2d4) [0xfffec17bea14]
 1  /opt/nvidia/holoscan/python/lib/holoscan/graphs/../../../../lib/libucs.so.0(+0x2aba4) [0xfffec17beba4]
 2  /opt/nvidia/holoscan/python/lib/holoscan/graphs/../../../../lib/libucs.so.0(+0x2af4c) [0xfffec17bef4c]
 3  linux-vdso.so.1(__kernel_rt_sigreturn+0) [0xffffa02197c0]
 4  python3() [0x5402b0]
 5  /opt/nvidia/holoscan/python/lib/holoscan/core/_core.cpython-38-aarch64-linux-gnu.so(+0x80a24) [0xffff9cc10a24]
 6  /opt/nvidia/holoscan/python/lib/holoscan/core/_core.cpython-38-aarch64-linux-gnu.so(+0xba180) [0xffff9cc4a180]
 7  /opt/nvidia/holoscan/python/lib/holoscan/core/_core.cpython-38-aarch64-linux-gnu.so(+0xcdadc) [0xffff9cc5dadc]
 8  /opt/nvidia/holoscan/python/lib/holoscan/graphs/../../../../lib/libholoscan_core.so.0(_ZN8holoscan3gxf10GXFWrapper4tickEv+0x234) [0xffff9dc8dc44]
 9  /opt/nvidia/holoscan/python/lib/holoscan/graphs/../../../../lib/libgxf_std.so(_ZN6nvidia3gxf14EntityExecutor10EntityItem11tickCodeletERKNS0_6HandleINS0_7CodeletEEE+0x498) [0xffff9d5b4008]
10  /opt/nvidia/holoscan/python/lib/holoscan/graphs/../../../../lib/libgxf_std.so(_ZN6nvidia3gxf14EntityExecutor10EntityItem4tickElPNS0_6RouterE+0x344) [0xffff9d5b4f14]
11  /opt/nvidia/holoscan/python/lib/holoscan/graphs/../../../../lib/libgxf_std.so(_ZN6nvidia3gxf14EntityExecutor10EntityItem7executeElPNS0_6RouterERl+0x2c0) [0xffff9d5b5750]
12  /opt/nvidia/holoscan/python/lib/holoscan/graphs/../../../../lib/libgxf_std.so(_ZN6nvidia3gxf14EntityExecutor13executeEntityEll+0x364) [0xffff9d5b5ea4]
13  /opt/nvidia/holoscan/python/lib/holoscan/graphs/../../../../lib/libgxf_std.so(+0xee79c) [0xffff9d53a79c]
14  /lib/aarch64-linux-gnu/libstdc++.so.6(+0xccfac) [0xffff9e8fcfac]
15  /lib/aarch64-linux-gnu/libpthread.so.0(+0x7624) [0xffff9fea5624]
16  /lib/aarch64-linux-gnu/libc.so.6(+0xd149c) [0xffff9ffa049c]
=================================
Segmentation fault (core dumped)
`

grlee77, 2023-11-17:
Hi @genevanmeter, it looks like this error was captured with version 0.6 of the SDK?

I cannot tell much from that traceback other than that something went wrong within the `compute` call of an operator (underlying GXF entity tick methods show up in the traceback).

It sounds like you have operators that use eCAL for publish/subscribe communication via shared memory? I don't know of other applications doing that, but it seems okay. 

If you could capture a log of the application with environment variable `HOLOSCAN_LOG_LEVEL=DEBUG` set, it might provide additional information. Depending on the confidentiality of the application, this may not be possible to post publicly as the log will have operator names, etc. I am mainly interested in what are the immediate lines preceding the segfault if it is not possible to provide a full log.

genevanmeter, 2023-11-17:
Yes holoscan v0.6.  
```
[2023-11-17 13:53:19.228][959][holoscan][debug][entity_executor.cpp:644] [C00007] tick codelet pixci
[2023-11-17 13:53:19.228][959][holoscan][debug][entity_executor.cpp:644] [C00007] tick codelet pixci
[2023-11-17 13:53:19.228][959][holoscan][debug][entity_executor.cpp:644] [C00007] tick codelet pixci
[2023-11-17 13:53:19.228][959][holoscan][debug][entity_executor.cpp:644] [C00007] tick codelet pixci
[2023-11-17 13:53:19.228][959][holoscan][debug][entity_executor.cpp:644] [C00007] tick codelet pixci
[2023-11-17 13:53:19.228][959][holoscan][debug][entity_executor.cpp:644] [C00007] tick codelet pixci
[2023-11-17 13:53:19.229][959][holoscan][debug][entity_executor.cpp:644] [C00007] tick codelet pixci
[2023-11-17 13:53:19.229][959][holoscan][debug][entity_executor.cpp:644] [C00007] tick codelet pixci
[2023-11-17 13:53:19.229][959][holoscan][debug][entity_executor.cpp:644] [C00007] tick codelet pixci
[2023-11-17 13:53:19.229][959][holoscan][debug][entity_executor.cpp:644] [C00013] tick codelet _broadcast_component_pixci_output
[2023-11-17 13:53:19.230][959][holoscan][debug][entity_executor.cpp:644] [C00017] tick codelet raw_output_local
[2023-11-17 13:53:19.230][959][holoscan][debug][trampolines.cpp:68] py_receive: Entity Case
[2023-11-17 13:53:19.230][959][holoscan][debug][trampolines.cpp:124] py_receive: TensorMap case
[2023-11-17 13:53:19.230][959][holoscan][debug][entity_executor.cpp:644] [C00025] tick codelet detectnet_detector
[2023-11-17 13:53:19.230][959][holoscan][debug][trampolines.cpp:68] py_receive: Entity Case
[2023-11-17 13:53:19.230][959][holoscan][debug][trampolines.cpp:124] py_receive: TensorMap case
[2023-11-17 13:53:19.234][959][holoscan][debug][entity_executor.cpp:644] [C00035] tick codelet raw_output
[2023-11-17 13:53:19.234][959][holoscan][debug][trampolines.cpp:68] py_receive: Entity Case
[2023-11-17 13:53:19.234][959][holoscan][debug][trampolines.cpp:124] py_receive: TensorMap case
[2023-11-17 13:53:19.234][959][holoscan][debug][entity_executor.cpp:644] [C00043] tick codelet mot
[2023-11-17 13:53:19.235][959][holoscan][debug][trampolines.cpp:262] py_receive: Python object case
[2023-11-17 13:53:19.237][959][holoscan][debug][entity_executor.cpp:644] [C00051] tick codelet merge
[2023-11-17 13:53:19.237][959][holoscan][debug][trampolines.cpp:262] py_receive: Python object case
[2023-11-17 13:53:19.238][959][holoscan][debug][entity_executor.cpp:644] [C00059] tick codelet box_viz
[2023-11-17 13:53:19.238][959][holoscan][debug][trampolines.cpp:68] py_receive: Entity Case
[2023-11-17 13:53:19.238][959][holoscan][debug][trampolines.cpp:124] py_receive: TensorMap case
[2023-11-17 13:53:19.238][959][holoscan][debug][trampolines.cpp:68] py_receive: Entity Case
[2023-11-17 13:53:19.238][959][holoscan][debug][trampolines.cpp:124] py_receive: TensorMap case
[helex-tracker-coarse:924  :0:959] Caught signal 11 (Segmentation fault: address not mapped to object at address 0xb8)
==== backtrace (tid:    959) ====
 0  /opt/nvidia/holoscan/python/lib/holoscan/graphs/../../../../lib/libucs.so.0(ucs_handle_error+0x2d4) [0xfffef413ea14]
 1  /opt/nvidia/holoscan/python/lib/holoscan/graphs/../../../../lib/libucs.so.0(+0x2aba4) [0xfffef413eba4]
 2  /opt/nvidia/holoscan/python/lib/holoscan/graphs/../../../../lib/libucs.so.0(+0x2af4c) [0xfffef413ef4c]
 3  linux-vdso.so.1(__kernel_rt_sigreturn+0) [0xffffbe0bf7c0]
 4  python3() [0x5402b0]
 5  /opt/nvidia/holoscan/python/lib/holoscan/core/_core.cpython-38-aarch64-linux-gnu.so(+0x80a6c) [0xffffbaab6a6c]
 6  /opt/nvidia/holoscan/python/lib/holoscan/core/_core.cpython-38-aarch64-linux-gnu.so(+0xba180) [0xffffbaaf0180]
 7  /opt/nvidia/holoscan/python/lib/holoscan/core/_core.cpython-38-aarch64-linux-gnu.so(+0xcdad0) [0xffffbab03ad0]
 8  /opt/nvidia/holoscan/python/lib/holoscan/graphs/../../../../lib/libholoscan_core.so.0(_ZN8holoscan3gxf10GXFWrapper4tickEv+0x234) [0xffffbbb33c44]
 9  /opt/nvidia/holoscan/python/lib/holoscan/graphs/../../../../lib/libgxf_std.so(_ZN6nvidia3gxf14EntityExecutor10EntityItem11tickCodeletERKNS0_6HandleINS0_7CodeletEEE+0x498) [0xffffbb45a008]
10  /opt/nvidia/holoscan/python/lib/holoscan/graphs/../../../../lib/libgxf_std.so(_ZN6nvidia3gxf14EntityExecutor10EntityItem4tickElPNS0_6RouterE+0x344) [0xffffbb45af14]
11  /opt/nvidia/holoscan/python/lib/holoscan/graphs/../../../../lib/libgxf_std.so(_ZN6nvidia3gxf14EntityExecutor10EntityItem7executeElPNS0_6RouterERl+0x2c0) [0xffffbb45b750]
12  /opt/nvidia/holoscan/python/lib/holoscan/graphs/../../../../lib/libgxf_std.so(_ZN6nvidia3gxf14EntityExecutor13executeEntityEll+0x364) [0xffffbb45bea4]
13  /opt/nvidia/holoscan/python/lib/holoscan/graphs/../../../../lib/libgxf_std.so(+0xee79c) [0xffffbb3e079c]
14  /lib/aarch64-linux-gnu/libstdc++.so.6(+0xccfac) [0xffffbc7a2fac]
15  /lib/aarch64-linux-gnu/libpthread.so.0(+0x7624) [0xffffbdd4b624]
16  /lib/aarch64-linux-gnu/libc.so.6(+0xd149c) [0xffffbde4649c]
=================================
Segmentation fault (core dumped)
```

genevanmeter, 2023-11-18:
By enabling eCAL to publish over the network and subscribing to a topic on a remote machine, I have a reliable and quick way to trigger the crash within a few seconds.  

It appears to be due to the subscription callback for incoming eCAL messages which I believe eCAL creates a thread for each subscription.  I initially thought to add a mutex but it did not have any effect.  If I remove the callback and instead receive the eCAL topic inside the `compute`  function it appears to solve my issue.    

So appears to be pointing to an eCAL issue.

closed 2023-12-15 by genevanmeter