nvidia-holoscan/holoscan-sdk #16: Unable to use VPI with igpu container

genevanmeter, 2023-11-20:
Holoscan 0.6
AGX Orin 64GB igpu
Jetpack 5.1.1
Running in docker with a display, no ssh

Trying to run VPI (https://docs.nvidia.com/vpi/index.html) in python with Holoscan

From the instructions on https://catalog.ngc.nvidia.com/orgs/nvidia/teams/clara-holoscan/containers/holoscan
```
export NGC_CONTAINER_IMAGE_PATH="nvcr.io/nvidia/clara-holoscan/holoscan:v0.6.0-igpu"
xhost +local:docker
nvidia_icd_json=$(find /usr/share /etc -path '*/vulkan/icd.d/nvidia_icd.json' -type f,l -print -quit 2>/dev/null | grep .) || (echo "nvidia_icd.json not found" >&2 && false)
docker run -it --rm --net host \
  --runtime=nvidia \
  -v /tmp/.X11-unix:/tmp/.X11-unix \
  -v $nvidia_icd_json:$nvidia_icd_json:ro \
  -e NVIDIA_DRIVER_CAPABILITIES=graphics,video,compute,utility,display \
  -e DISPLAY=$DISPLAY \
  --ipc=host \
  --cap-add=CAP_SYS_PTRACE \
  --ulimit memlock=-1 \
  --privileged \
  ${NGC_CONTAINER_IMAGE_PATH}
 ```
 From within the container:
 ```
apt update
apt install -y python3.8-vpi2
python3 -c "import vpi"
```
Expected Result:
```
root@ubuntu:/opt/nvidia/holoscan# python3 -c "import vpi"
root@ubuntu:/opt/nvidia/holoscan# 
```
Result:
```
root@ubuntu:/opt/nvidia/holoscan# python3 -c "import vpi"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ImportError: VPI_ERROR_INTERNAL: (EGL_BAD_PARAMETER) Failed to get EGL display
Segmentation fault (core dumped)
```
This does not occur on x86 with dgpu container. See https://docs.nvidia.com/vpi/installation.html
If the container is changed to `nvcr.io/nvidia/l4t-jetpack:r35.3.1` VPI will load successfully.

agirault, 2023-12-05:
Thanks for posting this here. We'll look into that ASAP and get back to you.

agirault, 2023-12-05:
What is your value for `DISPLAY` ?

>  If the container is changed to nvcr.io/nvidia/l4t-jetpack:r35.3.1 VPI will load successfully.

Does that include all the same runtime arguments?

genevanmeter, 2023-12-06:
>      What is your value for DISPLAY ?
```
root@ubuntu:/opt/nvidia/holoscan# echo $DISPLAY
:1
```

> If the container is changed to nvcr.io/nvidia/l4t-jetpack:r35.3.1 VPI will load successfully.
Does that include all the same runtime arguments?  

Yes it works with the same runtime arguments.

> Could you also add -e NVIDIA_VISIBLE_DEVICES=all even if it should already be the default for that img?

Same result
> What is the result of running grep libEGL -r /etc/nvidia-container-runtime/host-files-for-container.d on the host?

```
/etc/nvidia-container-runtime/host-files-for-container.d/l4t.csv:lib, /usr/lib/aarch64-linux-gnu/tegra-egl/libEGL_nvidia.so.0
```
> What is the result of running find / -name 'libEGL_nvidia.so*' 2>/dev/null in the container?
```
root@ubuntu:/opt/nvidia/holoscan# find / -name 'libEGL_nvidia.so*' 2>/dev/null
/usr/lib/aarch64-linux-gnu/tegra-egl/libEGL_nvidia.so.0
```

agirault, 2023-12-08:
Thank you. And for 3, the results are the same in both containers?

Couple more questions:
4. what is the result of `nvidia-ctk --version`? Could you try after upgrading it if older than 1.12 ?
5. what is the result of `grep X11 -r /etc/ssh/*config* | grep -v '#'` ? If X11 forwarding is on, can you disable it in the config file and restart sshd (`sudo systemctl restart ssh`) or just disable sshd (`sudo systemctl stop ssh && sudo systemctl disable ssh`) before retrying?

genevanmeter, 2023-12-13:
> Thank you. And for 3, the results are the same in both containers?
>
> Couple more questions: 4. what is the result of `nvidia-ctk --version`? Could you try after upgrading it if older than 1.12 ? 5. [...]

Yes both `nvcr.io/nvidia/l4t-jetpack:r35.3.1` and `nvcr.io/nvidia/clara-holoscan/holoscan:v0.6.0-igpu` report 
```
root@ubuntu:/opt/nvidia/holoscan# find / -name 'libEGL_nvidia.so*' 2>/dev/null
/usr/lib/aarch64-linux-gnu/tegra-egl/libEGL_nvidia.so.0
```

```
nvidia-ctk --version
NVIDIA Container Toolkit CLI version 1.11.0-rc.1
commit: 629a689
```
```
grep X11 -r /etc/ssh/*config* | grep -v '#'
/etc/ssh/sshd_config:X11Forwarding yes
```
I disabled sshd and got the same error as before.  

I have also reflashed the Jetson to Jetpack 5.1.1 with the same error.  

I have refactored our dockerfiles to use `nvcr.io/nvidia/l4t-jetpack:r35.3.1` to continue development.

agirault, 2024-03-01:
Can you report if you're still seeing an issue using v1.0.3?

genevanmeter, 2024-03-05:
> Can you report if you're still seeing an issue using v1.0.3?

I tried the `nvcr.io/nvidia/clara-holoscan/holoscan:v1.0.3-igpu` container on an updated Jetpack 6 Orin AGX.  The docker image does not appear to contain l4t apt sources (bug?) which has the updated python3.10-vpi3 package.  I added the repo manually and I was then able to install python3.10-vpi3 and run the test successfully.

Once in the container:
```
apt-key adv --fetch-key https://repo.download.nvidia.com/jetson/jetson-ota-public.asc
echo "deb https://repo.download.nvidia.com/jetson/common r36.2 main
deb https://repo.download.nvidia.com/jetson/t234 r36.2 main" > /etc/apt/sources.list.d/nvidia-l4t-apt-source.list
apt update
apt install -y python3.10-vpi3
```
Quick test:
```
python3 -c "import vpi"
```

agirault, 2024-03-18:
Great @genevanmeter ! 

I am not convinced that it makes sense to have the apt source files in the container since some repos might vary depending on which iGPU device you're running on (IGX, Jetson AGX) and that's not exactly portable, so adding it yourself or mounting the one from your host makes sense to me. 

I'll close this issue since your VPI problem is addressed, but please create another one if you think the source repository is another problem that should be addressed as well, as my first gut feeling might not be adequate.

closed 2024-03-18 by agirault