nvidia-holoscan/holoscan-sdk #22: VideoStreamRecorderOp fails create *.gxf_index file on the flight

jpmazellier, 2024-03-20:
I posted a question on the NVIDIA developer forum about this issue: [(https://forums.developer.nvidia.com/t/videostreamrecorderop-issue-with-gxf-index-empty/285639)]
Basically, the *.gxf_entities file is well populated by the operator VideoStreamRecorderOp on the flight, but not the *.gxf_index file (empty) until the application closes "properly". The only way i found was to have a "proper" application closing (using a VideoStreamReplayerOp with a fixed number of frames as source video). 
So in case of crash or forced interruption, the video stream can not be reconstructed.
Would it be possible to have the *.gxf_index file populated regularly so to have a chance to recover the video even in case of application crash ? 
Thanks !

tbirdso, 2024-03-20:
Hi @jpmazellier , thank you for reporting this issue and documenting your experience in the Holoscan forum! We will look into what it would take to update `VideoStreamReplayerOp` behavior as you've described.

tbirdso, 2024-03-21:
Hi @jpmazellier , after investigation we've determined that this feature is currently available in `VideoStreamRecorderOp` in Holoscan v1.0.3. Please set the `flush_on_tick` option to `True` when initializing your `VideoStreamRecorderOp` in order to flush both the binary and index streams on each tick. Note that you may observe a performance impact with this feature enabled.

I will close this issue as already implemented, but please re-open it if you continue to observe issues with `flush_on_tick` enabled. If you have further questions on operator parameters or `VideoStreamRecorderOp` please add them to your forum thread and we will follow up there.

Thank you for using Holoscan!

closed 2024-03-21 by tbirdso