nvidia-holoscan/holoscan-sdk #24: Support for more Resolutions and Framerates with AJA

genevanmeter, 2024-04-17:
Feature Request

The AJASourceOp is very limited in supported resolutions and framerates, only supports 1080p60 vs 2160p60

https://github.com/nvidia-holoscan/holoscan-sdk/blob/cf61c8e44a37d4f4bd8f4650fe7fe537249ea05e/src/operators/aja_source/aja_source.cpp#L90-L99

It also seems to only support HDMI or SDI input, no analog.

https://github.com/nvidia-holoscan/holoscan-sdk/blob/cf61c8e44a37d4f4bd8f4650fe7fe537249ea05e/src/operators/aja_source/aja_source.cpp#L173-L178

tbirdso, 2024-05-02:
Hi, thank you for your request! We have taken this feature request under consideration and plan to expand `AJASourceOp` support for further resolutions and framerates in a Holoscan SDK release around the start of July (currently milestone v2.2).

tbirdso, 2024-09-05:
The following formats are supported in `AJASourceOp` as of Holoscan SDK v2.4.0 [(release notes)](https://github.com/nvidia-holoscan/holoscan-sdk/releases/tag/v2.4.0):
- 720p @ 50, 59.94, 60Hz
- 1080i @ 50, 59.94, 60Hz
- 1080p @ 23.98, 24, 25, 29.97, 30, 50, 59.94, 60Hz
- 3840x2160 (UHD) @ 23.98, 24, 25, 29.97, 30, 50, 59.94, 60Hz
- 4096x2160 (4K) @ 23.98, 24, 25, 29.97, 30, 50, 59.94, 60Hz

closed 2024-09-05 by tbirdso