nvidia-holoscan/holoscan-sdk #27: any major issue is using C# for developing holodscan application

man-chi, 2024-06-08:
Is there any success story using C# (.NET) to build a medical imaging application based on the Holoscan SDK?

I am aware that Holoscan SDK does support C++ and Python. 

since my existing software engineers are based on C# skillsets, there is a roadmap for having our medical imaging application cross-platform, supporting both embedded Nvidia IGX devices and Nvidia AI microservices in the cloud. 

if my team decides to use C# to develop Holoscan, what are the potential challenges, risks, and disadvantages to take note of? 

please kindly advise!

tbirdso, 2024-06-10:
Hi @man-chi , Holoscan SDK supports C++ and Python on Linux amd64 and arm64 platforms. If you are using C#/.NET on Linux, you could explore calling into the Holoscan SDK C++ API from your C# application.

man-chi, 2024-06-11:
thanks, I am exploring dotnet's c++/CLI to call holoscan c++ SDK.
https://learn.microsoft.com/en-us/cpp/dotnet/dotnet-programming-with-cpp-cli-visual-cpp?view=msvc-170

have you heard of any open-source project or success store that is using a similar approach? I would like to connect to other developers and learn any major issues and limitations that Holoscan may not work properly.

tbirdso, 2024-08-01:
Hi @man-chi , I'd like to close this issue but feel free to continue discussion on your [forum post](https://forums.developer.nvidia.com/t/any-major-issue-if-using-c-for-developing-holodscan-application/295919) on the subject. Holoscan SDK currently supports C++ and Python distributions, C# is not currently on our roadmap.