nvidia-holoscan/holoscan-sdk #28: pip is not able to install holoscan 2.0.0

vikashg, 2024-06-12:
I am trying to install holoscan 2.0.0 for monai-deploy-app-sdk. But issuing the command `pip install holoscan` installs version 0.6.0. 
Even if I try `pip install holoscan==2.0.0`, I get the following error
```
ERROR: Could not find a version that satisfies the requirement holoscan==2.0.0 (from versions: 0.0.0.post1, 0.4.0, 0.4.1, 0.5.0, 0.5.1, 0.6.0)
ERROR: No matching distribution found for holoscan==2.0.0
```
For context the behaviour is same if I use a conda environment or a pip environment.
My python version is 3.10 and pip version is 24 and my os is Ubuntu 20.04.

Related issue and context: https://github.com/Project-MONAI/monai-deploy-app-sdk/issues/492
Adding @MMelQin for visibility.

tbirdso, 2024-06-12:
Hi @vikashg , thank you for sharing your environment information. Starting in [release v1.0.3](https://github.com/nvidia-holoscan/holoscan-sdk/releases/tag/v1.0.3) Holoscan SDK requires Ubuntu >=22.04 (glibc>=2.35) to build or install. We've documented supported platforms in the [user guide](https://docs.nvidia.com/holoscan/sdk-user-guide/sdk_installation.html#sd-tab-item-2).

The following options are available:
1. Update your machine to use Ubuntu 22.04
2. Continue to use Holoscan SDK v0.6.0
3. Try to build Holoscan SDK wheels from [source](https://github.com/nvidia-holoscan/holoscan-sdk) for your environment. However, this approach is not maintained..

MMelQin, 2024-06-12:
> Hi @vikashg , thank you for sharing your environment information. Starting in [release v1.0.3](https://github.com/nvidia-holoscan/holoscan-sdk/releases/tag/v1.0.3) Holoscan SDK requires Ubuntu [...]

Thanks @tbirdso, the OS version could be the one, even though the Pypi package name says anylinux. I had upgraded my host OS before developing and using Holoscan 2.x

vikashg, 2024-06-12:
Thanks @tbirdso, I feel its time to upgrade the OS.. :)

tbirdso, 2024-06-12:
> the OS version could be the one, even though the Pypi package name says anylinux.

@MMelQin  the "manylinux" name comes from the [manylinux](https://github.com/pypa/manylinux) project for distributing wheels across Linux distros, we're following the standard naming convention there for PyPI. The `manylinux_2_35` tag is what's important to indicate glibc >= 2.35 is required.

@vikashg I'll go ahead and close this issue, but please reopen or file a new issue if you aren't able to install after your update.

MMelQin, 2024-06-12:
> @MMelQin the "manylinux" name comes from the [manylinux](https://github.com/pypa/manylinux) project for distributing wheels across Linux distros, we're following the standard naming convention there [...]

Yes, the package file does not matter as much, as metadata says "manylinux: glibc 2.35+ x86-64" which I believe gets the pip (on Ubuntu 20.04) to eliminate holoscan v1.0+ packages.