nvidia-holoscan/holoscan-sdk #29: holoviz render_buffer_output bug ?

jpmazellier, 2024-06-14:
Dear Devs,

I'm trying to use the render_buffer_output from Holoviz operator, but I systematically get an error : 
[error] [entity.hpp:90] Unable to find component from the name 'render_buffer_output' (error code: 24)

To elucidate the root cause, I started from the following python script : holoscan-sdk/examples/holoviz/python/holoviz_geometry.py
I create a simple sink operator that has an "input_message" port. The compute method just print the received messages (see code snippet below): 
```python
class SinkOp(Operator):
    """Sink operator which consumes the input messages and display the content.

    This operator has:
        inputs:  "input_message"
    """

    def __init__(
        self,
        fragment,
        *args,
        **kwargs,
    ) -> None:
        """
        Constructor of the SinkOp class.
        """
        # Need to call the base class constructor last
        super().__init__(fragment, *args, **kwargs)

    def setup(self, spec: OperatorSpec):
        spec.input("input_message")

    def compute(self, op_input, op_output, context):
        """
        This function is called on each tick of SinkOp.
        """
        # receiving input stream tensor
        in_message = op_input.receive("input_tensor")
        print(f"{self.name}: Received message: {in_message}")
        return None
``` 

I then modify declaration of holoviz operator by adding ```enable_render_buffer_output=True``` and I add a connection between my sink operator and holoviz (using the enabled "render_buffer_output" port) : 
```python
self.add_flow(visualizer, sink, {("render_buffer_output", "input_message")})
```

The holoviz operator works fine for displaying the dynamic geometries on top of racerx video, but I continuously get the following message: 
```
[error] [entity.hpp:90] Unable to find component from the name 'render_buffer_output' (error code: 24)
sink: Received message: {'render_buffer_output': None}
```
Same result if I switch parameter ```headless=True``` in operator definition: no Holoviz window (as expected) but same error message as above.

Only solution I can find so far is to insert a FormatConverterOp in between my HolovizOp (port "render_buffer_output") and a consumer operator. Going this way, I can use the render buffer (after conversion) as retrieved from the dict key "" of the message from port "tensor" of the format converter. The declaration of the FormatConverterOp that works here is:
```python
        allocator = UnboundedAllocator(self, name="pool")

        converter = FormatConverterOp(
            fragment=self,
            name="converter",
            pool=allocator,  # mandatory to assign for FormatConverterOp
            in_dtype="rgba8888",
            out_dtype="rgb888", 
        )
```

Is this a bug or do I miss something ? Maybe declaring a input buffer is necessary but I don't think so following documentation that mentions about named output ```render_buffer_output```: "If an input render buffer is specified, it is using that one, else it allocates a new buffer."
It's still fine to use a FormatConverterOp but maybe this should be mentioned in the doc if this is no bug. 
Thanks for your help here :)

tbirdso, 2024-06-17:
Hi @jpmazellier , thanks for reporting your experience. From discussion with @AndreasHeumann this is known behavior. `HolovizOp` `render_buffer_output` outputs a `gxf::VideoBuffer` object which is currently only supported among C++ operators, while `FormatConverter` outputs a `Tensor` object that is supported in both C++ and Python.

We plan to investigate `gxf::VideoBuffer` support in Python. A workaround for the time being is to convert `gxf::VideoBuffer` to `Tensor` output in a C++ operator as you've done with `FormatConverter`.

jpmazellier, 2024-06-18:
OK, thank you very. This is also the workaround I found to be able to use the buffer output. 
Thanks for you help, 
Best

closed 2024-06-18 by jpmazellier