nvidia-holoscan/holoscan-sdk #32: bayer_demosaic operator incorrectly expects RGB/RGBA input image

jaydcarlson, 2024-07-25:
In holoscan SDK 2.1.0, code was added in to ensure the input image has rank 3:
```
auto in_rank = in_tensor_gxf.rank();
if (in_rank != 3) {
    throw std::runtime_error(
       fmt::format("Input tensor has {} dimensions. Expected a tensor with 3 dimensions "
                   "(corresponding to an RGB or RGBA image).",
                   in_rank));
}
```
But this makes no sense. The input to a demosaic operator is a single-channel bayer image (rank 2).

Consequently, this check breaks the entirety of Holoscan Sensor Bridge, which relies on the demosaic operator in every example.

tbirdso, 2024-07-26:
Hi @jaydcarlson , thank you for reporting this issue. Could you please confirm the version of Holoscan Sensor Bridge that you are using for reference?

From discussion with @AndreasHeumann , the rank check here is consistent with other tensor handling in Holoscan such as in Holoviz. Please consider the following workarounds:
- Use Holoscan Sensor Bridge 1.1-EA, which is compatible with Holoscan SDK v2.1.0
- Downgrade to Holoscan SDK v2.0.0 and use Holoscan Sensor Bridge 1.0

grlee77, 2024-07-26:
Thank @jaydcarlson. I agree that it is confusing that a 3rd dimension (which I think must be size 1) is required. I can open an issue to update the operator to also accept 2D input in future releases (and still allow 3D as long the channels dimension is size 1)

tbirdso, 2024-09-05:
Addressed in [Holoscan SDK v2.4.0](https://github.com/nvidia-holoscan/holoscan-sdk/releases/tag/v2.4.0).

closed 2024-09-05 by tbirdso