nvidia-holoscan/holoscan-sdk #34: ninja: error: 'lib/libajantv2d.a', needed by 'lib/libholoscan_op_aja.so.2.3.0', missing and no known rule to make it

sese1815, 2024-08-30:
## Please describe your question
Ask about Holoscan SDK features or best practices. For example, "Does Holoscan SDK support v4l2 compatible cameras?"
 This is a question and not a bug report since i am not sure if i just did a mistake or it is real issue. i followed the steps for building Holoscan-SDK. Due to some errors i changed the html from 2.0 to 1.1. 

Now i get ninja: error: 'lib/libajantv2d.a', needed by 'lib/libholoscan_op_aja.so.2.3.0', missing and no known rule to make it whenever i do ./build run --target debug. I am not sure if those two problems are related and i also don't know how to fully restart. As yo
## Please specify what Holoscan SDK version you are using
latest

## Please add any details about your platform or use case
I am using a Linux (Ubuntu 22.04.4) Cuda 12.2 

I am not that experienced so i would really appreciate your help and thoughts on my Problem.

tbirdso, 2024-08-30:
Hi @sese1815 , thank you for the report. The issue you've observed is a known issue in Holoscan SDK 2.3.0 when building in `Debug` mode, and is scheduled for resolution in Holoscan SDK v2.4.0.

As a workaround, you can either:
- Build in `Release` mode: `./run build --target release`
- Manually set the ajantv2 build type to `Debug` at https://github.com/nvidia-holoscan/holoscan-sdk/blob/main/cmake/deps/ajantv2_rapids.cmake#L37 and build in `Debug` mode: `./run build --target debug`

tbirdso, 2024-09-06:
Fixed in [Holoscan SDK v2.4.0](https://github.com/nvidia-holoscan/holoscan-sdk/releases/tag/v2.4.0).

closed 2024-09-06 by tbirdso