nvidia-holoscan/holoscan-sdk #36: Dynamic configuration approach

JohnMoon-VTS, 2024-11-05:
Does Holoscan have a way to dynamically update state in an operator based on some external command?

For example, my team is developing a Holoscan application with the following layout (PR [here](https://github.com/nvidia-holoscan/holohub/pull/572)):

```
┌───────────┐    ┌─────┐    ┌─────┐    ┌─────────────┐    ┌──────────┐ VRT
│ File Read ├───►│ FFT ├───►│ PSD ├───►│PSD Averaging├───►│Packetizer├─────►
├───────────┤    └─────┘    └─────┘    └─────────────┘    └──────────┘
│Raw IQ Data│
└───────────┘
```

Now, say we wanted to present an external interface (maybe gRPC) to modify some variables (such as file sample rate or FFT size). My initial thought is to do something like this:

```
                                                                            
┌───────────┐    ┌─────┐    ┌─────┐    ┌─────────────┐    ┌──────────┐ VRT  
│ File Read ├───►│ FFT ├───►│ PSD ├───►│PSD Averaging├───►│Packetizer├─────►
├───────────◄──┐ └──▲──┘    └─────┘    └─────────────┘    └──────────┘      
│Raw IQ Data│  │    │                                                       
└───────────┘  │    │                                                       
               │    │                                                       
               │    │   ┌─────────────┐                                     
               └────┴───┼ gRPC Server │                                     
                        └─────────────┘                                     
```

So, the gRPC server would field requests from outside of Holoscan and then go twiddle values in the relevant operators.

In GNU Radio, this sort of approach would be supported with [Message Passing](https://wiki.gnuradio.org/index.php/Message_Passing), but it seems like in Holoscan this would need to be done with Ports. Is there any concept of message passing in Holoscan (i.e. trigger a callback when I get some message in)? It seems like adding a "configuration" port and checking for new config in every `compute()` run isn't the right approach.

JohnMoon-VTS, 2024-11-05:
Another thought: I could start the gRPC server as a background thread at the application level and pass it handles to the different operators. Then, it could just modify the config values directly instead of needing a messaging interface. I'm not sure how this would work on a multi-host configuration though.

tbirdso, 2025-01-16:
Hi @JohnMoon-VTS , checking on this issue following close of  https://github.com/nvidia-holoscan/holohub/pull/572.  There was some offline discussion when this was posted, did you find the guidance you needed?

JohnMoon-VTS, 2025-01-16:
Hi @tbirdso, I think the approach we've outlined in the comment above will work for us, though it does feel somewhat hacky. We're open to other suggestions if there's a "first class" way of doing this in Holoscan.

To expand on our strategy a bit, in our application's `main()`, before running the Holoscan app, we kick off a thread which launches a gRPC server and pass it a handle to the `app` object. Like:

```c++
    // Run
    app->is_metadata_enabled(true);
    app->config(config_path);

    // Kick off the background thread
    std::thread t(&PsdPipelineSim::update_thread, app);

    app->run();
    app->run_thread = false;
    t.join();
    return 0;
```

Then, the `update_thread` takes in gRPC messages and updates the operators it has as member variables.
```c++
// on gRPC message
this.fftOp->set_fft_size(message.fft_size);
```

So, it's not inter-operator message passing, but just the application fielding messages and manipulating its operators from the top down.

If this seems like the right way of doing what we're trying to do, feel free to close this issue!

tbirdso, 2025-01-16:
The approach you've shared seems valid. If you're interested in an operator-centric approach you might review the [GRPC-based Endoscopy Tool Tracking demo application](https://github.com/nvidia-holoscan/holohub/tree/main/applications/distributed/grpc/grpc_endoscopy_tool_tracking) on HoloHub. The demo app "cloud" process similarly runs both the GRPC server and Holoscan application instances, but also uses a dedicated receiver operator to ingest messages from GRPC into the Holoscan SDK pipeline for data propagation via port connections.

Closing for now but please feel free to re-open if you have more questions.

closed 2025-01-16 by tbirdso