nvidia-holoscan/holoscan-sdk #39: Glfw Error 65550: Failed to detect any supported platform

suveerudayashankara, 2024-12-10:
[info] [fragment.cpp:588] Loading extensions from configs...
[info] [gxf_executor.cpp:262] Creating context
[info] [gxf_executor.cpp:1764] creating input IOSpec named 'receivers:0'
[info] [gxf_executor.cpp:1764] creating input IOSpec named 'render_buffer_input'
[info] [gxf_executor.cpp:1764] creating input IOSpec named 'camera_up_input'
[info] [gxf_executor.cpp:1764] creating input IOSpec named 'camera_look_at_input'
[info] [gxf_executor.cpp:1764] creating input IOSpec named 'camera_eye_input'
[info] [gxf_executor.cpp:1764] creating input IOSpec named 'input_specs'
[info] [gxf_executor.cpp:1764] creating input IOSpec named 'receivers'
[info] [gxf_executor.cpp:2171] Activating Graph...
[info] [gxf_executor.cpp:2201] Running Graph...
[info] [gxf_executor.cpp:2203] Waiting for completion...
[info] [greedy_scheduler.cpp:191] Scheduling 2 entities
Authorization required, but no authorization protocol specified
Glfw Error 65550: Failed to detect any supported platform
[error] [gxf_wrapper.cpp:59] Exception occurred when starting operator: 'visualizer' - Failed to initialize glfw
[warning] [entity_executor.cpp:539] Failed to start entity [visualizer]
[warning] [greedy_scheduler.cpp:243] Error while executing entity 12 named 'visualizer': GXF_FAILURE
[error] [entity_executor.cpp:630] Entity [visualizer] must be in Started, Tick Pending, Ticking or Idle stage before stopping. Current state is StartPending
[info] [greedy_scheduler.cpp:401] Scheduler finished.
[error] [program.cpp:580] wait failed. Deactivating...
[error] [runtime.cpp:1649] Graph wait failed with error: GXF_FAILURE
[warning] [gxf_executor.cpp:2204] GXF call GxfGraphWait(context) in line 2204 of file /workspace/holoscan-sdk/src/core/executors/gxf/gxf_executor.cpp failed with 'GXF_FAILURE' (1)
[info] [gxf_executor.cpp:2214] Graph execution finished.
[error] [gxf_executor.cpp:2222] Graph execution error: GXF_FAILURE
Traceback (most recent call last):
  File "/opt/nvidia/holoscan/examples/v4l2_camera/python/v4l2_camera.py", line 95, in <module>
    main(config_file=config_file)
  File "/opt/nvidia/holoscan/examples/v4l2_camera/python/v4l2_camera.py", line 89, in main
    app.run()
RuntimeError: Failed to initialize glfw
[info] [gxf_executor.cpp:291] Destroying context
[image]

USING rtx 4090 for the work
but i am facing problems related to GXF and video replayer related program.

but Holoscan is running well

genevanmeter, 2024-12-12:
You'll likely need to enable X11 or Wayland forwarding.

```
 To leverage a display, add the flags below:
    X11 :
        run xhost +local:docker as a prerequisite so that X11 is configured to allow commands from docker
        add -v /tmp/.X11-unix:/tmp/.X11-unix -e DISPLAY to the docker run command
   Wayland: add -e ${XDG_RUNTIME_DIR} -v ${XDG_RUNTIME_DIR}::${XDG_RUNTIME_DIR} -e XDG_SESSION_TYPE -e WAYLAND_DISPLAY to the docker run command
```
Source: https://catalog.ngc.nvidia.com/orgs/nvidia/teams/clara-holoscan/containers/holoscan
Just after the docker run it explains some options you can set.

suveerudayashankara, 2024-12-14:
It works now thank you very much

agirault, 2024-12-14:
Thank you for the inquiry @suveerudayashankara and for the support @genevanmeter !

> ./dev_container launch --ssh_x11
so it gave all the permissions for it to work

@genevanmeter are you saying it worked once you passed --ssh_x11 ? Or that you expected permissions to work since you used it and it did not?