nvidia-holoscan/holoscan-sdk #40: Output ConditionType.NONE op execution with Event/MultiThread Scheduler

genevanmeter, 2024-12-12:
## Please describe your question
Trying to understand where I should put ConditionType.NONE to create a hot path and cold path.  In the graph below I would want:
Hot Path: tx -> mx -> rx
Cold Path: tx -> mx -> slowop
[image]

Initially it would make sense to put in on tx so it would always push the latest data even if downrange wasn't ready.  What was observed was slowop would prevent mx from sending the data to rx so the pipeline is dropping tx data.  It is not clear why rx was blocked from executing with a multithread/event scheduler.
 
Putting ConditionType.NONE on both tx and mx or just on mx does what I am expecting.

## Please specify what Holoscan SDK version you are using
2.7.0

## Please add any details about your platform or use case
x86_64 dGPU

```
from holoscan.conditions import CountCondition
from holoscan.core import Application, ConditionType
from holoscan.operators import PingRxOp, PingTxOp
from holoscan.decorator import create_op, Output
from holoscan.schedulers import EventBasedScheduler, GreedyScheduler, MultiThreadScheduler
from time import sleep

x = 0
@create_op(
    # outputs=Output("out")
    outputs=Output("out", condition_type=ConditionType.NONE, condition_kwargs={})
)
def tx_op():
    global x
    x += 1
    sleep(0.1)
    return x

@create_op(
    inputs="val_in",
    outputs=Output("out")
    # outputs=Output("out", condition_type=ConditionType.NONE, condition_kwargs={})
)
def mx_op(val_in):
    return val_in + 1

@create_op(
    inputs="val_in",
)
def slow_op(val_in):
    sleep(0.5)
    print(f"Slow op got {val_in}")

class MyPingApp(Application):
    def compose(self):

        tx = tx_op(self, CountCondition(self, 20), name="tx")
        mx = mx_op(self, name="mx_op")
        rx = PingRxOp(self, name="rx")
        slowop = slow_op(self, name="slow_op")
        

        # Define the workflow:  tx -> mx -> rx (hot path)
        #                              | -> slowop  (cold path)                 
        self.add_flow(tx,mx)
        self.add_flow(mx, rx)
        self.add_flow(mx, slowop)

def main():
    app = MyPingApp()
    event_based = True
    scheduler_class = EventBasedScheduler if event_based else MultiThreadScheduler
    scheduler = scheduler_class(
        app,
        worker_thread_number=4,
        stop_on_deadlock=True,
        stop_on_deadlock_timeout=500,
        name="multithread_scheduler",
    )
    app.scheduler(scheduler)
    app.run()

if __name__ == "__main__":
    main()
```

grlee77, 2024-12-12:
Hi @genevanmeter,

I think the reason is that by default holoscan places the following condition on operator ports (setting the `ConditionType.NONE` condition on the port would override this for a given port)
- each input port will have a `MessageAvailableCondition` that prevents the operator from executing if no message is available on the receiver
- each output port will have a `DownstreamMessageAffordableCondition` that prevents the operator from executing if **all** downstream operators do not have space in their receiver queues.

In your example, you need the NONE condition on the output of `mx` operator when the two branches operate at different rates. Otherwise, the slower `slowop` queue will be full and prevent `mx` from running again. 

You could also potentially change `mx` to have two separate output ports (that each output the same data) and only put the NONE condition on the one connecting to the slow path if you want to also still ensure you don't transmit messages to `rx` faster than it can process them.

I am not sure if you already saw it, but there is also one example in the SDK with a similar usage scenario of two branches operating at different temporal rates in:
https://github.com/nvidia-holoscan/holoscan-sdk/tree/main/examples/multi_branch_pipeline

genevanmeter, 2024-12-14:
Thank you for the explanation.

closed 2024-12-14 by genevanmeter