nvidia-holoscan/holoscan-sdk #42: SDK build failure: test video clips no longer downloadable

jpeeee, 2025-01-15:
**Describe the bug**
The video clips downloaded with racerx_data are no longer available at URI https://api.ngc.nvidia.com/v2/resources/nvidia/clara-holoscan/holoscan_racerx_video/versions/20231009/zip.
As a result the build fails

**To Reproduce**
Steps to reproduce the behavior:
1. Try to build holoscan-sdk
2. CMake will fail to configure has it cannot download the package

or browse to https://api.ngc.nvidia.com/v2/resources/nvidia/clara-holoscan/holoscan_racerx_video/versions/20231009/zip. and see  error 401 GONE.

**Expected behavior**
The racex.zip package should be downloadable

**Screenshots**

**Platform Information (please complete the following information):**
 - Holoscan SDK Version [e.g. 2.8.0]
- Architecture: [arm64]
 - OS: [Ubuntu]

**Additional context**

wendell-hom, 2025-01-15:
@jpeeee there's been a behavior change in NGC.  We are aware of this issue and will fix this.  In the meantime, do you have the `data/` folder downloaded from a copy of the repo that you can copy over?  If the files are already there, then the build step skips the download and avoids the issue as a temporary workaround.

wendell-hom, 2025-01-15:
Alternatively, you can download the data using ngc cli with this command.  Please use an up-to-date version of ngc cli that you can download from https://org.ngc.nvidia.com/setup/installers/cli and run.

ngc registry resource download-version "nvidia/clara-holoscan/holoscan_racerx_video:20231009"

Once the data is downloaded, you can move it to the `data/racerx` folder

tbirdso, 2025-01-30:
Resolved in Holoscan SDK v2.9. Please update to Holoscan SDK >=v2.9 and use NGC CLI to retrieve resource data archives from NGC.

For developers relying on Holoscan SDK <=v2.8:
- Please install the latest NGC CLI [here](https://org.ngc.nvidia.com/setup/installers/cli)
- Refer to this HoloHub commit for how to update `holoscan_download_data` calls to use NGC CLI rather than HTTPS
- Open a GitHub issue or post on the [Holoscan forum](https://forums.developer.nvidia.com/c/healthcare/holoscan-sdk/320/none) if you continue to experience difficulty.

closed 2025-01-30 by tbirdso