nvidia-holoscan/holoscan-sdk #44: Dynamic flow behaviour different when using HolovizOp operator

moo-joshua, 2025-03-19:
## Please describe your question
Hi all, I'm currently trying to implement the newly introduced dynamic flow within my workflow but I am encountering some issues where only the first frame is displayed and the scheduler stops with the following message:
`[info] [greedy_scheduler.cpp:372] Scheduler stopped: Some entities are waiting for execution, but there are no periodic or async entities to get out of the deadlock.`

Here's a short snippet of my code and the structure of the pipeline, note that MotionDetectorOp, PreprocessOp and PostprocessOp are custom operators. Basically, I'd only like to run inference on frames with motion, and if there's no motion I'd only want the MotionDetectorOp to emit the frame and displayed by Holoviz.:
```
class MyApp(application):
         ......

  def compose(self):
      allocator = UnboundedAllocator(self, name="allocator")
  
      source = VideoStreamReplayerOp(
          self, name="replayer", directory=self.video_dir, **self.kwargs("replayer")
      )
  
      motion_detector = MotionDectectorOp(
          self, name="motiondetector", sensitivity=0.2, resolution=self.resolution, **self.kwargs("motiondetector")
      )
  
      preprocessor = FormatConverterOp(
          self, name="preprocessor", pool=allocator, **self.kwargs("preprocessor")
      )
  
      inference = InferenceOp(
          self,
          name="inference",
          allocator=allocator,
          model_path_map=self.model_path_map,
          **self.kwargs("inference"),
      )
      
      inference_preprocessor = PreprocessOp(self, name="preprocessor2", **self.kwargs("preprocessor2"))
  
      postprocessor = PostprocessOp(self, name="postprocessor", **self.kwargs("postprocessor"))
  
      viz = HolovizOp(self, name="viz", **self.kwargs("viz"))
  
      # Define the workflow
      self.add_flow(source, viz, {("output", "receivers")})
      self.add_flow(source, motion_detector, {("output","input_tensor")})
      self.add_flow(motion_detector, preprocessor, {("output", "source_video")})
      self.add_flow(preprocessor, inference_preprocessor, {("tensor", "input_tensor")})
      self.add_flow(inference_preprocessor, inference, {("output","receivers")})
      self.add_flow(inference, postprocessor, {("transmitter","input_tensor")})
      self.add_flow(postprocessor, viz, {("output", "receivers")})
      self.add_flow(motion_detector, viz, {("output", "receivers")})
      
      
      # Set dynamic flow based on condition
      def route_flow(op):
          if op.motion_flag:
              print('going to preprocessor')
              op.add_dynamic_flow(preprocessor) 
          else:
              print('going to holoviz')
              op.add_dynamic_flow(viz,"receivers") 
      
      self.set_dynamic_flows(motion_detector, route_flow)
```
[image]

Upon replacing the HolovizOp with just a simple Sink operator I wrote, the dynamic workflow works as intended:

```
class Sink(Operator):

    def __init__(self, fragment, *args, **kwargs):
        super().__init__(fragment, *args, **kwargs)

    def setup(self, spec:OperatorSpec):
        spec.input("receivers")
        # spec.output("output")

    def compute(self, op_input, op_output, context):
        x = op_input.receive("receivers")
        print('Inside Sink')
```
where viz is now replaced with sink:
```
      # Define the workflow
      self.add_flow(source, sink, {("output", "receivers")})
      self.add_flow(source, motion_detector, {("output","input_tensor")})
      self.add_flow(motion_detector, preprocessor, {("output", "source_video")})
      self.add_flow(preprocessor, inference_preprocessor, {("tensor", "input_tensor")})
      self.add_flow(inference_preprocessor, inference, {("output","receivers")})
      self.add_flow(inference, postprocessor, {("transmitter","input_tensor")})
      self.add_flow(postprocessor, sink, {("output", "receivers")})
      self.add_flow(motion_detector, sink, {("output", "receivers")})
      
      
      # Set dynamic flow based on condition
      def route_flow(op):
          if op.motion_flag:
              print('going to preprocessor')
              op.add_dynamic_flow(preprocessor)  
          else:
              print('going to sink')
              op.add_dynamic_flow(sink,"receivers")  
      
      self.set_dynamic_flows(motion_detector, route_flow)
```

Can anyone explain why there is a difference in behaviour? Could it be that the HolovizOp interacts differently to the scheduler compared to a custom operator like my Sink?

 
## Please specify what Holoscan SDK version you are using
latest (v3.0.0)

## Please add any details about your platform or use case
For instance, "AGX Orin"

moo-joshua, 2025-03-20:
Hi @gigony, thanks for your reply. Unfortunately I think I'm still don't quite understand how to use delegates. I understand that InferenceOp and HolovizOp essentially requires constantly an message at the inputs in order to prevent the deadlock. Are the delegate operators essentially providing a "dummy/placeholder" input to InferenceOp/HolovizOp? How would I specify the "dummy/placeholder" inputs if the operators prior to that are not selected based on the dynamic flow callback? For example: if `op.motion_flag = False`, this means that the flow would be source -> motion_detector -> holoviz_delegate -> viz, what is happening at the inference_preprocessor -> inference_delegate level if preprocessor wasn't run in the first place?

 I've modified my code to the following but I'm unsure if its correct, am I meant to change the add_dynamic_flow lines in the callback function to the delegate operators as well? Seems like I've now done something wrong by adding a dynamic flow to viz but I've not added the flow in the first place: 

```python
 # Define the workflow
        self.add_flow(source, viz, {("output", "receivers")})
        self.add_flow(source, motion_detector, {("output", "input_tensor")})
        self.add_flow(motion_detector, preprocessor, {("output", "source_video")})
        self.add_flow(preprocessor, inference_preprocessor, {("tensor", "input_tensor")})
        # self.add_flow(inference_preprocessor, inference, {("output","receivers")})
        self.add_flow(inference, postprocessor, {("transmitter","input_tensor")})
        # self.add_flow(postprocessor, viz, {("output", "receivers")})
        # self.add_flow(motion_detector, viz, {("output","receivers")})

        self.add_flow(postprocessor, holoviz_delegate, {("output", "in")})
        self.add_flow(motion_detector, holoviz_delegate, {("output", "in")})
        self.add_flow(inference_preprocessor, inference_delegate, {("output", "in")})

        self.add_flow(holoviz_delegate, viz, {("out", "receivers")})
        self.add_flow(inference_delegate, inference, {("out","receivers")})

        # Set dynamic flow based on condition
        def route_flow(op):
            if op.motion_flag:
                print('going to preprocessor')
                op.add_dynamic_flow('output',preprocessor, "source_video") 
            else:
                print('going to sink')
                op.add_dynamic_flow('output',viz, "receivers")

        self.set_dynamic_flows(motion_detector, route_flow)
```

gigony, 2025-03-27:
@moo-joshua I apologize for the delayed response.

> Are the delegate operators essentially providing a "dummy/placeholder" input to InferenceOp/HolovizOp?

The primary purpose of the delegate operators is to accept a single input from multiple upstream operators and forward it to a single input port of the target operator (`viz`).

In the workflow, we have three possible paths:

- Path 1: source → viz (static/constant path)
- Path 2: source → motion_detector → holoviz_delegate → viz (when `op.motion_flag = False`)
- Path 3: source → motion_detector → preprocessor → inference_preprocessor → inference → post_processor → holoviz_delegate → viz (when `op.motion_flag = True`)

Given the dynamic nature of the flow, `holoviz_delegate` may receive data from either:
- Path 2: the `output` port of `motion_detector`
- Path 3: the `output` port of `post_processor`

Its output is then routed to a single input port of `viz`.
(Note: `holoviz_delegate` has only one input port.)

Please refer to the diagrams below:

1) Without `holoviz_delegate`  
[image]

2) With `holoviz_delegate`  
[image]

With `holoviz_delegate`, the input to `viz` is always populated (via path 1 and either path 2 or path 3).

Since only a single output port (`inference_preprocessor`) connects to the `receivers` multi-input port of `inference`, we don’t need an `inference_delegate`.

Your workflow could be updated as follows:
```python
# Define the workflow
self.add_flow(source, viz, {("output", "receivers")})
self.add_flow(source, motion_detector, {("output", "input_tensor")})
self.add_flow(motion_detector, preprocessor, {("output", "source_video")})
self.add_flow(preprocessor, inference_preprocessor, {("tensor", "input_tensor")})
self.add_flow(inference_preprocessor, inference, {("output", "receivers")})
self.add_flow(inference, postprocessor, {("transmitter", "input_tensor")})
self.add_flow(postprocessor, viz, {("output", "receivers")})
self.add_flow(motion_detector, viz, {("output", "receivers")})

self.add_flow(postprocessor, holoviz_delegate, {("output", "in")})
self.add_flow(motion_detector, holoviz_delegate, {("output", "in")})
# self.add_flow(inference_preprocessor, inference_delegate, {("output", "in")})

self.add_flow(holoviz_delegate, viz, {("out", "receivers")})
# self.add_flow(inference_delegate, inference, {("out", "receivers")})

# Set dynamic flow based on condition
def route_flow(op):
    if op.motion_flag:
        print('going to preprocessor')
        op.add_dynamic_flow('output', preprocessor, "source_video")
    else:
        print('going to sink')
        op.add_dynamic_flow('output', holoviz_delegate, "in")  # modified to route to holoviz_delegate when motion_flag is False

self.set_dynamic_flows(motion_detector, route_flow)
```
Please let me know if above suggestion work.

Thank you!

closed 2025-03-27 by moo-joshua