nvidia-holoscan/holoscan-sdk #46: Holoviz Fails to Create Vulkan Instance in Docker + WSL2 (RTX 3080) — No Video Output

rithwikjayanth233, 2025-04-23:
Hi,
I'm running the yolo_model_deployment sample from HoloHub using Holoscan SDK v3.1.0 inside a Docker container on WSL2 (Ubuntu 22.04) with an RTX 3080 GPU. The pipeline runs fine with excellent inference performance, but the app fails when trying to create the HolovizOp due to a Vulkan initialization error.

`[error] [gxf_wrapper.cpp:63] Exception occurred when starting operator: 'detection_visualizer' - Failed to create the Vulkan instance.
'
'
[warning] [gxf_executor.cpp:2429] GXF call GxfGraphWait(context) failed with 'GXF_FAILURE'
'
'
[error] [program.cpp:580] wait failed. Deactivating...
`

Even after installing vulkan-tools in the container:
`
vulkaninfo
'
'
ERROR: vkCreateInstance failed with ERROR_INCOMPATIBLE_DRIVER
`

Environment:

- Windows 11
- WSL2 (Ubuntu 22.04)
- Docker v28.0.4
- NVIDIA driver v576.02 (CUDA 12.9)
- GPU: RTX 3080
- Holoscan SDK v3.1.0
- Running Holohub's yolo_model_deployment sample

I tried setting $DISPLAY, confirmed /tmp/.X11-unix/X0 exists, and even installed vulkan-tools inside the container, but vulkaninfo fails with the same vkCreateInstance error.

Is there any official guidance or known workaround for running Holoviz on WSL2 + Docker?

Thanks in advance!

AndreasHeumann, 2025-05-16:
Hi,

There are two issues here.

First there seems to be a problem with with the Vulkan setup. You can use `export VK_LOADER_DEBUG=all` to get the log of the Vulkan loader, also see https://vulkan.lunarg.com/doc/sdk/1.4.313.0/windows/LoaderDebugging.html for more information.

Secondly, Miscrosoft is using a Vulkan to Direct3D wrapper driver on the WSL2 system, there is no NVIDIA Vulkan driver available. The Holoviz operator of the Holoscan SDK requires the NVIDIA Vulkan driver since it needs extensions to efficiently support rendering CUDA memory content with Vulkan.

I hope this helps.

tbirdso, 2025-07-01:
Hi, closing as not supported following @AndreasHeumann 's response above.