nvidia-holoscan/holoscan-sdk #47: How to deal with buffered input video frames for Holoviz operator

moo-joshua, 2025-05-01:
## Please describe your question
Hi, I am currently trying to implement a point tracking algorithm which requires an initial N frames to provide a track for each point. Here's a simple schematic: 

[image]

So far, I've managed to set up Point-Tracker to only run when there are N number of frames in its input port's queue/buffer by doing:

```
class PointTrackerOp(Operator):
        def __init__(...):
                 ...

        def setup(self, spec: OperatorSpec):
               spec.input("in", size=IOSpec.IOSize(8))
               spec.output("out")

        def compute(self, op_input, op_output, context):
               ......
               out_message = {'points', point_coords} #point_coords are 2d coordinates of the points across the N frames.
               op_output.emit(out_message, "out")
               
```
 
Such that `compute` will process those N frames and outputs a track `point_coords` [N x 2]

However, I'm not sure how to set up the Holoviz operator to display each frame with the tracked points and make sure the coordinates for the points across the N frames are synced with the frames coming in from the source. I suspect I would have to also buffer the holoviz input? But there's still an issue of displaying each frame in the buffer one at a time.

Is there something like a timestamp for each message received such that the visuals can be synced?

## Please specify what Holoscan SDK version you are using
v3.0.0

## Please add any details about your platform or use case
For instance, "AGX Orin"

gigony, 2025-05-16:
Thanks @moo-joshua for raising the issue!

The `ExecutionThrottlerOp` in the following example ([streaming_with_monitor](https://github.com/nvidia-holoscan/holoscan-sdk/blob/main/examples/flow_control/streaming_with_monitor)) might help you design a throttling mechanism that accepts input and overwrites the previous input in the output queue if a downstream (slower) operator is still processing. This allows the pipeline to proceed without blocking other faster operators.

- https://github.com/nvidia-holoscan/holoscan-sdk/blob/main/examples/flow_control/streaming_with_monitor/python/streaming_with_monitor.py#L84-L89

> class ExecutionThrottlerOp(Operator):  
>     """Controls the flow of messages between operators with different processing rates."""  
>     def setup(self, spec: OperatorSpec):  
>         spec.input("input")  
>         # Set the queue policy to POP to avoid push failures when emitting a message to the output  
>         # port. Also, set the condition to NONE so that the operator executes immediately upon  
>         # receiving input, even if the downstream operator is still busy (e.g., due to a long  
>         # computation in detect event operator).  
>         **spec.output("output", policy=IOSpec.QueuePolicy.POP).condition(ConditionType.NONE)**  
>     def compute(self, op_input, op_output, context):  
>         value = op_input.receive("input")  
>         if value is not None:  
>             op_output.emit(value, "output")

Regarding timestamp support, ensure the application uses a multi-threaded scheduler like `EventBasedScheduler` (by default, the single-threaded `GreedyScheduler` is used):
- https://github.com/nvidia-holoscan/holoscan-sdk/blob/main/examples/flow_control/streaming_with_monitor/python/streaming_with_monitor.py#L241-L242

Although the `emit()` method accepts a timestamp for internal use ([example](https://github.com/nvidia-holoscan/holoscan-sdk/blob/main/examples/conditions/expiring_message/python/ping_expiring_message.py#L50-L63)), there isn’t a straightforward way to access this timestamp in Python currently.

Instead, refer to:
- **[Working With Metadata from Operator::compute](https://docs.nvidia.com/holoscan/sdk-user-guide/holoscan_create_app.html#working-with-metadata-from-operator-compute)**
- [Example usage in test_operator_metadata_apps.py](https://github.com/nvidia-holoscan/holoscan-sdk/blob/main/python/tests/system/test_operator_metadata_apps.py#L63)

I recommend setting `ConditionType.NONE` on the upstream operator’s output port and using the Metadata API to attach a timestamp (or frame ID) before emitting data (in an upstream operator such as ExecutionThrottlerOp operator in [stream_with_monitor app](https://github.com/nvidia-holoscan/holoscan-sdk/blob/main/examples/flow_control/streaming_with_monitor/python/streaming_with_monitor.py#L84-L89)). In the downstream operator, you can retrieve this metadata after calling `op_input.receive()`.

Also, you may need to buffer the input to Holoviz if you want to display or synchronize N frames before passing them in.

If you intend to display all N frames in `HoloVizOp`, refer to this example:
- https://github.com/nvidia-holoscan/holoscan-sdk/blob/main/examples/holoviz/python/holoviz_views.py

[image]

moo-joshua, 2025-05-20:
Thanks for the information, will need some time to go through these concepts! Will reopen if anymore questions arise!

closed 2025-05-20 by moo-joshua