nvidia-holoscan/holoscan-sdk #51: Inconsistent Image and Geometry Layer Drawing with InputSpec.View

genevanmeter, 2025-07-19:
**Describe the bug**
When using InputSpec.View with geometry items (rectanges, triangles, etc.) and a matrix it appears to be applied differently than the image layer.  It seems image layer is [0...1] and the geometry layer is [-1...1].  However, if the matrix is not used it appears to be applied as [0...1]  

**To Reproduce**
Adding a rectangle [(0,0),(1,1)] and apply the same manipulations as the upper right image to be rotated along with an image.  I modified holoviz_view.py to demostrate below.

**Expected behavior**
A purple box drawn encapulating the entire rotating upper right image.  What is observed is the box in the lower right corner of the rotated image.

**Screenshots**
Incorrect Positioning

[image]

Disabling the matrix for the purple box and its where I would expect.

[image]

**Platform Information (please complete the following information):**
 - Holoscan SDK Version 3.4.0
- Architecture: x86_64
 - OS: Ubuntu 24.04

**Additional context**
Add any other context about the problem here.
```
"""
SPDX-FileCopyrightText: Copyright (c) 2023-2024 NVIDIA CORPORATION & AFFILIATES. All rights reserved.
SPDX-License-Identifier: Apache-2.0

Licensed under the Apache License, Version 2.0 (the "License");
you may not use this file except in compliance with the License.
You may obtain a copy of the License at

http://www.apache.org/licenses/LICENSE-2.0

Unless required by applicable law or agreed to in writing, software
distributed under the License is distributed on an "AS IS" BASIS,
WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
See the License for the specific language governing permissions and
limitations under the License.
"""  # noqa: E501

import math
import os
from argparse import ArgumentParser

import numpy as np

from holoscan.core import Application, Operator, OperatorSpec
from holoscan.operators import HolovizOp, VideoStreamReplayerOp
from holoscan.resources import CudaStreamPool

sample_data_path = os.environ.get("HOLOSCAN_INPUT_PATH", "../data")

class Matrix4x4:
    """4x4 row major matrix

    Note : as_...() means that the whole matrix is being modified
    set_...() only changes the concerned fields of the matrix

    """

    def __init__(self):
        self.identity()

    def identity(self):
        self.matrix_ = np.eye(4)

    def data(self):
[... 297 lines cut ...]
                max_size=5,
            ),
        )
        self.add_flow(source, image_views)
        self.add_flow(image_views, visualizer, {("outputs", "receivers")})
        self.add_flow(image_views, visualizer, {("output_specs", "input_specs")})

def main(config_count):
    app = MyVideoViewsApp(config_count=config_count)
    app.run()

if __name__ == "__main__":
    # Parse args
    parser = ArgumentParser(description="Example video processing application")
    parser.add_argument(
        "-c",
        "--count",
        default=0,
        help="Set the number of frames to display the video",
    )
    args = parser.parse_args()

    main(config_count=args.count)

```

AndreasHeumann, 2025-07-21:
Hi Gene,

thanks for your detailed report.

You are right, as long as no matrix is applied the geometry layer coordinate range is [0...1]. This has historical reasons since Holoviz started as a 2D renderer and matrix support and 3D rendering had been added later in the process.

This is also described int the [documentation](https://docs.nvidia.com/holoscan/sdk-user-guide/api/cpp/function_modules_2holoviz_2src_2holoviz_2holoviz_8hpp_1a016553789904105ab21b356226e2e650.html#exhale-function-modules-2holoviz-2src-2holoviz-2holoviz-8hpp-1a016553789904105ab21b356226e2e650):

`Layers can also be placed in 3D space by specifying a 3D transformation matrix. Note that for geometry layers there is a default matrix which allows coordinates in the range of [0 … 1] instead of the Vulkan [-1 … 1] range. When specifying a matrix for a geometry layer, this default matrix is overwritten.`.

You can workaround this inconsistency by always specifying geometry coordinates in the [-1...1] range and always using a view matrix (use the identity matrix if there is no transformation).

I hope this helps, let me know if you have more questions.

Best regards,
Andreas

genevanmeter, 2025-07-22:
Thanks for the breakdown.

closed 2025-07-22 by genevanmeter